SimonMerrett/SOICbite #2: Prevent reverted connection

X-Ryl669, 2019-06-13:
If, instead of the 5 holes after the contact pin, you fill one of them (for example, the 2nd one to the top) and cut the matching plastic part on the grip, you have revert connection protection.
It's even probably cheaper since that make one less hole to drill on the PCB.

This worth it if you intend to reprogram your device few month later and don't want to spend time figuring out the pinout of the SOICbite's connection, or if you have (unskilled) workers programming board in series.

SimonMerrett, 2019-06-14:
If you look in the side profile of the SOIC clip I modified in the [project photos](https://github.com/SimonMerrett/SOICbite/blob/master/images/slot_cutout_clip_side.jpeg), you'll see that the spring conductors of the clip are closer to the PCB, so I don't think this keyed design would save you from VCC touching the wrong pads but visually it might help you line stuff up. 

There is a silkscreen half loop around pad 1, which can be lined up with the red index wire coming out of the SOIC clip, or you could even mark the SOIC clip plastic itself.

FWIW, I unplug my IDC connector on the other end of the ribbon cable, then connect the SOICbite and then replug the IDC in. 

I'm happy to keep this issue open if you want to submit a version that is keyed, otherwise I'll close. Please let me know.

X-Ryl669, 2019-06-14:
Not sure I'm getting what you are saying. In the file SOICbite_connected.jpg, it seems like you've printed a plastic part that enclose the clip to target the 5 holes in the PCB. I thought you could print 4 pins instead of 5 (in the layout like this: ` o   o o o`) and have the same in the PCB. If you don't have a plastic part enclosing the clip, then I'm confused about the use of the 5 holes in the PCB. What are they for ?

SimonMerrett, 2019-06-14:
I haven't printed any plastic parts, just used an off-the-shelf SOIC-8 test clip and sliced some of the jaws in the centre so it will close tight enough for the clip's metal (sprung) contacts to touch the PCB pads.
The holes in the PCB are for the small claws on the SOIC clip to grip onto. This is for alignment and mechanical hold, but the reality is with at least the SOIC clip that _I have_ is that the metal contacts will touch the PCB even if there were no holes for the claws to sink into. So although your idea would give you some indication that you were correctly orientated (if incorrect, one claw would be holding each side of the jaw clear of the PCB and the clip could easily be pulled off, in theory), I don't think it would stop you from accidentally connecting pads to contacts the wrong way around. Would love a solution to this but the current suggestions of removing holes and claws doesn't seem to achieve this _with the gear that I have_. YMMV, so please try and let me know. There is an advantage of removing some holes and claws, which is that you can keep thicker traces to escape the pads.

SimonMerrett, 2019-06-15:
Hi @X-Ryl669 closing now as no reply - feel free to re-open if needed.

closed 2019-06-15 by SimonMerrett

X-Ryl669, 2020-06-16:
I've made some change to your solution [here](https://github.com/X-Ryl669/SOICconnect). I'm printing a small plastic part that's ensuring the position and prevent inverted connection. It's also helping for connection since it gives a physical feedback with some border to push the PCB edge onto and a fang to laterally center.

If you have a 3D printer, it's a 10mn print time that'll save your board from shifted pin connections.