SimonMerrett/SOICbite #3: SWD pinout

rewolff, 2019-06-16:
I'm an STM32 user, so I use the SWD connector. (not yet this project). 

I am not aware of the "SWO" pin in that configuration.... 

The "discovery" boards also have the target NRST signal connected on the programming connector. there is a spot available for that -> do it.... (possibly mark as optional... )

The "discovery" boards have a 6-pin connector and I have been unable to find what the 6th was ever intended for. 
I have now repurposed that for the "boot0" signal. With a pulldown to GND and a switch to VCC I can then boot the target into bootloader mode. 

I suggest to mark one of the further free pins for that purpose. Then, I'd be able to have two of the clamps, one on my programmer and one to a two-switch-board that allows me to hit the boot,  and reset buttons to make the target switch to bootloader mode. (=DFU over USB for most of my targets).

SimonMerrett, 2019-06-17:
That'd be fine by me - please submit a pr so I can see exactly what you want.

rewolff, 2019-06-17:
Will do. Creating eagle layout now.

SimonMerrett, 2019-06-17:
Thanks. Will you leave the pin assignments on the eagle footprint unlabelled so that it can be used for any pin configuration? Would prefer to leave pin assignment conventions in the Readme.md for now. Fantastic to have an eagle footprint though - much appreciated.

rewolff, 2019-06-17:
Eagle has "sub-devices" that can have different footprints. e.g. for a soic/dip version of a chip. I was hoping to do it the other way around, but alas, that's not possible. So for now I have a "generic soicbite" that in the symbol simply exports pin 1, pin2 etc.. 
Later: application specific schematic symbols that export the pinout for a specific CPU so that it's easy to hookup correctly for that specific CPU/family.

SimonMerrett, 2019-06-17:
@rewolff , does https://github.com/SimonMerrett/SOICbite/blob/master/SOICbite_Eagle.lbr help? Thanks to @VoswijkC

closed 2019-06-17 by SimonMerrett