SimonMerrett/SOICbite #6: Converter boards

DefProc, 2019-07-23:
Do you have thoughts on generating converter boards that go from the SOIC clip standard IDC connectors to protocol specific interfaces eg. typical USB/UART module pinout? 

I'm aware that this might move SOICbite away from a more connection agnostic solution, but the table of suggested connection pins in the Readme was really good from my point of view, it lets me just take the connection I want to add, and route to a pad arrangement that already exists, instead of having to create new ones each time, eg. [xkcd.com/927](https://xkcd.com/927/).

I've forked a repo at [DefProc/SOICbite](https://github.com/DefProc/SOICbite/tree/master/converters) with a USB converter I'm trying as an example.

SimonMerrett, 2019-07-29:
@DefProc sorry not to have replied sooner. I do this kind of board for my programmers already but I really like your USB version! Have you built/tested it yet? I was particularly pleased with the 2MHz default SWD clock with my Atmel ICE working with a relatively bodged and elongated cable/connector arrangement last week.

SimonMerrett, 2019-07-29:
@DefProc but to more specifically answer your question:

1. My first thought is that making converter/adapter boards is a good idea that compliments the SOICbite concept.
2. How to add value? Perhaps including good pinout labelling on the silkscreen? Maybe a voltage regulator in some cases (I'm thinking how the FTDI boards don't have sufficient current from 3.3V pin for the ESP range of chips). Good mechanical design with mounting holes as a minimum for "nailing" to the bench but perhaps even a second layer of PCB that acts as a "lid" held on standoffs, with cutouts for the connectors. Possibly the option to have a separate power supply (jumper bridge to select source) to the IC, rather than e.g. the USB 5V pin? Could be good for testing lower voltages etc?

DefProc, 2019-07-31:
@SimonMerrett I've got some [boards in the post from Aisler](https://aisler.net/p/CMPNOGUJ), I'll let you know how testing goes. 

[image]

Yes for good pin labelling, I really like it when boards can be self documenting, even if it means taking a little more space. Doubly for development boards.

It feels like it's worth adding a feature if it solves a problem that you have when using an interface normally. Although I'm a bit fan of the Unix philosiphy of "do one thing well"; if the IDE connector is standard enough that it could be reasonable to apply a specific function to a specialist connection board (eg. [USB power injector](https://www.tindie.com/products/mux/usb-datapower-merger-anti-splitter/)) rather than trying to have one board to all possible jobs for an interface.

Do you have a preferred standard for spacing/sizing of mounting holes? M3 clearance holes on a metric spaced grid? Are you trying to "nail" it other existing hardware?

Is it worth selecting some standard measurements to allow anyone to expand from based on need?

SimonMerrett, 2019-08-01:
> Do you have a preferred standard for spacing/sizing of mounting holes? M3 clearance holes on a metric spaced grid? Are you trying to "nail" it other existing hardware?

I think M3 makes sense as a very common standoff dimension. I'd go so far to suggest we ought to seek multiples of 2mm or 5mm as the grid pitch. I'm not attaching to anything specific - it's just something we would wished we had included if we omit it! But I don't think this is necessary to use SOICbite - just suggestions if you don't have a better starting point. As you say, I think self-documenting boards/pins are the most important bit. That way there is minimum likelihood of magic smoke release if someone picks up and connects the wrong adapter board.

Nice board - I like the "bite" in the outline!

SimonMerrett, 2019-08-01:
So if your test goes well with this board, shall we have an "adapter board" folder in the main repo?

DefProc, 2019-08-05:
Sure, that's all working as expected. I'll open a pull request to add the boards in a subfolder.

SimonMerrett, 2019-08-12:
Hi, sorry for the delay. I'm getting back into things this week after holiday. Will get to the pr as soon as I can.

SimonMerrett, 2019-08-14:
Commit #7 should let us close this for now.

closed 2019-08-14 by SimonMerrett