SimonMerrett/SOICbite #12: 'Known good' SOIC Clip

joshtyler, 2020-02-02:
I've designed this connector into a PCB, I think it's quite a neat solution - thank you for designing it.

Unfortunately the SOIC clip I used was a bit flimsy, and so never made very good contact with the PCB, and before long the plastic guide pins broke.

I was wondering therefore if you had the model number of a SOIC clip that was easy to modify, and worked reliably?

SimonMerrett, 2020-02-02:
@joshtyler I'm afraid I have just been lucky with the cheapest of the cheap from eBay/Aliexpress etc. Let's keep this issue open for a while and see if anyone else has any tips for clips they'd recommend as it would be good to have one people could rely on a bit more.

whinis, 2020-04-29:
I have been using this one https://smile.amazon.com/Signstek-SOIC8-Socket-Adpter-Programmer/dp/B00V9QNAC4. Requires modifying as normal but clips fairly strongly. You can remove the center pin and stretch the spring a bit for an even strong clip

v01d, 2020-05-17:
Have you thought about designing something for 3D printing? Would mostly require an equivalent spring and contacts (maybe it could be designed for standard pin headers, after removing their plastic housing). This way the teeth could be made to grab the PCB before the pads are contacted

janmg, 2021-04-06:
The cheap black soic8 clamps seem to use weak plastic, so they can be used ten to twenty times but then their spring will crush the guides. I got mine for $0.70, so next time I need them, I'll invest in these more rigid looking test clips, I haven't tried them yet though.

CPT-063 Test Clip SOIC8 Pomona 5250
https://www.pomonaelectronics.com/products/test-clips/soic-clip-8-pin

SimonMerrett, 2021-04-06:
> but then their spring will crush the guides

Interesting - I haven't experienced this but not done many tens of use cycles. Please do update us with what you find when you get the other clips.

SimonMerrett, 2021-04-06:
> Have you thought about designing something for 3D printing?

No but open to suggestions. There are quite a few decent clothes-peg type custom clips out there, often built from PCB/Acrylic and pogo pins.

2nOrderEDO, 2024-09-25:
Hello Simon, thanks for the SOICbite idea, it really is very cool.

I had the chance to test it some days ago. At the beginning I faced problems with the clips not making contact with the pads. It significantly improved after stretching the spring out to increase the clipping force. Nevertheless, this did not solve all issues.

After careful inspection, I noticed the metal pins on my SOIC clip were very slightly out of the plastic. I also confirmed this was the case when comparing my clip with yours [here](https://github.com/SimonMerrett/SOICbite/blob/master/HOWTO_mod_clip.md)

It turns out it is possible to move in and out the pins by applying some force with pliers. After aligning the contact point of the pins over the pads, the clip is working flawlessly.

I propose to add a note like below to the documentation:

[image]

Regards,
2nOrderEDO