SimonMerrett/SOICbite #13: No license

davidlenfesty, 2020-03-20:
Thanks for putting this out there :).

I'm putting this into my personal KiCAD library, which I plan on throwing up on github, but I would feel weird putting it up without some sort of attribution, or confirmation that that is okay with you in some way.

It's really not necessary but I do suggest putting a license file somewhere, may I suggest BSD, or the [WTFPL](http://www.wtfpl.net/).

SimonMerrett, 2020-03-20:
@davidlenfesty you're right and I just couldn't be bothered to read the differences and choose one when I set up the repo.
It's certainly OK for you to include in your publicly hosted kicad library. Attribution would be appreciated but I don't consider it necessary as I haven't given anyone else who has built converter boards etc the impression that they ought to attribute. 
The only licence clause I could think is relevant is the usual disclaimer about being fit for purpose and provided as-is with no guarantee etc. If you know one that covers that (and perhaps also attribution if you can't find one standalone) please let me know which one it is and I'll add it. But otherwise crack on!

davidlenfesty, 2020-03-28:
The [2-clause BSD license](https://opensource.org/licenses/BSD-2-Clause) is the closest and simplest thing I could find.

It does require attribution but that just means copying the license file essentially.

SimonMerrett, 2020-03-28:
@davidlenfesty thanks for the research. License added!

closed 2020-03-28 by SimonMerrett