SimonMerrett/SOICbite #15: Flex restrictions

edoput, 2020-04-14:
Hello and thank you for this nice project.

I'm thinking of doing a flex PCB for running some UART lines out of a device case but
I'm stuck on how to later connect to the extension from a UART-USB adapter.

This looks promising as it is dead easy and does not require me to place a new connector (SMT or TH), hell I could reuse any SOIC clip lying around.

Let's say we use [OSH Park](https://docs.oshpark.com/services/flex/) and work along some limits.

Obviously we need solder mask on both side of the PCB but that is covered

>  Soldermask Type | SMOBC (Soldermask Over Bare Copper), both sides

We also need two layers of copper and that is also covered.

So the real question is if we can actually find a SOIC connector that can clamp this
extremely thin PCB as the stackup is the following.

| Layer | Thickness |
| -- | -- |
Silkscreen | 1mil (0.0254mm) | +/-0.2mil (0.00508mm)
Soldermask | 1mil (0.0254mm) | +/-0.2mil (0.00508mm)
Top Copper | 1oz  (1.4mil, 35um) |  
Polyimide Flexible Substrate | 4mil (0.1016mm) | +-0.4mil (0.0102mm
Bottom Copper | 1oz  (1.4mil, 35um) |  
Soldermask | 1mil (0.0254mm) | +/-0.2mil (0.00508mm)
Silkscreen | 1mil (0.0254mm) | +/-0.2mil (0.00508mm)

Which would yeld a total of 10.8 mil ( ~ 0.25 mm).

So maybe just one side would be available at a time, the other covered by some *thickening* to make the connection more reliable as otherwise the film might just slip out all the times.

We also have some restrictions on how far from the edge copper should be as the process
of cutting it's different from rigid material.

> *Board Edge Keepout* Copper within 15mil (0.381mm) of a board edge will be fabbed,  but may be damaged during shipping, copper within 6mil (0.152mm) of a board edge will be  removed prior to fabrication

So overall if the connector is not placed very close to the border it should survive the fabrication process and the SOIC clip can still reach it.

I'm interested in the drill holes as for [flex pcb there are different limits](https://docs.oshpark.com/services/flex/#drill-specifications) and the wear and tear might be different.

Is there any other physical requirement for this connector that I haven't gone over?

SimonMerrett, 2020-04-14:
[image]
Hey @edoput , from a quick and dirty look, you will need to cut away the plastic in the cyan box in the image above (the green represents the spring contacts, for contrast), which everyone needs to do unless using *very* thick PCB substrate. I also think in your case, you may need to cut away some plastic in the magenta box. Looking at the "claws" in the orange box and thinking about your requirements, you may want to cut away one of each opposing pair of claws, so that they interleave and still lock onto the holes, overall. You could just cut away one side of claws but this may not line the pads up with the spring contacts evenly. 

If I leave this issue open, will you try something out and let us know what works?

edoput, 2020-04-14:
I didn't know it would generally require modifications to the clip, good to know.

I'm looking for more off the shelf solutions (just running a flex cable with more pins that I need) but if
I ever get around to designing my flex pcb I'll put it on and see if it works.

SimonMerrett, 2020-04-16:
I'll close the issue then. Feel free to reopen if you have anything to report.

closed 2020-04-16 by SimonMerrett