SimonMerrett/SOICbite #16: Recommanded distance to board edge

X-Ryl669, 2020-06-02:
I'm finally trying to implement this footprint on one of my board, and I'm wondering what is the best expected distance from board edge ?

In the connector footprint, there's an inverted T on the bottom, but I just wanted to be sure it's meaning it's supposed to match the board edge:
[image]

Can you confirm ?

SimonMerrett, 2020-06-02:
Yes, that lower "baseline" in the image is supposed to be a guide to line up with the board edge. You could place it further into the board (limited by the "throat" of your clip) but this is probably the best compromise between being unobtrusive and relatively stable when the clip is connected.

SimonMerrett, 2020-06-02:
[image]

X-Ryl669, 2020-06-03:
Thanks! Very useful.

closed 2020-06-03 by SimonMerrett