SimonMerrett/SOICbite #18: SWD: 5V pin?

dbrgn, 2020-08-23:
I'm creating SOICbit footprints and components for https://librepcb.org/, in order to connect an ARM based board to an SWD debug probe (JLink).

JLink offers a (non-standard?) 5V pin on the connector (power from USB) that can be turned on via software. Do you think we could put it on pin 2 or 3, to get some kind of standardization?

SimonMerrett, 2020-08-23:
I have no problem with that. I should say that the suggestions on the Readme.md are not standards, merely suggestions. But I also appreciate that suggestions can become standards through convention and inertia. Will you submit an amended Readme.md with the PR for the Librepcb footprint files? I acknowledge that Librepcb is less dependent on Github repos for footprints - but it would be nice to host them here too.

closed 2020-08-23 by SimonMerrett

dbrgn, 2020-08-23:
Yes, I can open a PR!

LibrePCB libraries should usually be repositories themselves (for easier checkout), but I can submit a file with a link to the library repo (https://github.com/dbrgn/SOICbite.lplib).

SimonMerrett, 2020-08-23:
thanks for the link. I'm not an expert on licenses but you seem to have a different one than the one used for SOICbite. Is that compatible?

dbrgn, 2020-08-24:
CC0 is essentially public domain, so it's compatible with any license! It means "no restrictions or guarantees whatsoever".

SimonMerrett, 2020-08-24:
But no attribution. It's interesting as I don't really think copyright comes into play if you use a different file format to generate the same footprint in copper etc and you have linked to the original repo so there is kind of attribution there. But you don't impose attribution on anyone who copies your code. I'm not bothered about attribution to a large degree - the main reason I added a license was because someone asked for it. Just interesting to see how a license does or does not affect things! :-)

dbrgn, 2020-08-24:
CC0 is the recommendation for all LibrePCB libraries, since LibrePCB should be usable for commercial projects as well, and a project always contains the library elements. If you want to do a commercial project and include a library element that requires attribution, dealing with this in a legally correct way is probably quite tricky.

It seems that most people simply ignore this issue when using licenses with attribution requirements :slightly_smiling_face: